STMicroelectronics/stm32f0xx-hal-driver #4: USB HID Keyboard stopped working on STM32Cube FW_F0 V1.11.3

DeckerCHAN, 2021-09-29:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * Custom board with STM32F042K6U6
 * STM32CubeMX + Clion

**Describe the bug**
A clear and concise description of what the bug is.
Custome board simulate keyboard keep pressing CapLock preventing PC from entering sleep.

Programming following this post https://controllerstech.com/how-to-use-stm32-as-a-keyboard/

It was working all good by utilising STM32Cude FW_F0 V1.11.2 but stopped working on V1.11.3

The key was pressed but never released. 

**How To Reproduce**
1. Indicate the global behavior of your application project
```C
      keyboardhid[3] = 0x39;
      USBD_HID_SendReport(&hUsbDeviceFS, (uint8_t *) &keyboardhid, sizeof(keyboardhid));
      HAL_Delay(50);
      keyboardhid[3] = 0x00;
      USBD_HID_SendReport(&hUsbDeviceFS, (uint8_t *) &keyboardhid, sizeof(keyboardhid));
      HAL_Delay(1000);
```
2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
Recent updates on stm32f0xx_hal_pcd.c and stm32f0xx_ll_usb.c
3. The use case that generates the problem
https://controllerstech.com/how-to-use-stm32-as-a-keyboard/
4. How we can reproduce the problem
Just follow the post above

I will utilise a logic analyser to compare traffic between the two versions.  

I can provide a video but don't know where I can upload it.

ALABSTM, 2022-03-04:
Hi @DeckerCHAN,

Would you mind running one of these two applications from version 1.11.3 of the official STM32CubeF0 firmware on your board and check whether you can reproduce the issue you reported? Thank you in advance for your reply.
* [STM32F072B-Discovery#HID_Standalone](https://github.com/STMicroelectronics/STM32CubeF0/tree/master/Projects/STM32F072B-Discovery/Applications/USB_Device/HID_Standalone)
* [STM32072B_EVAL#HID_Standalone](https://github.com/STMicroelectronics/STM32CubeF0/tree/master/Projects/STM32072B_EVAL/Applications/USB_Device/HID_Standalone)

There should be no difficulty in porting them as both F072 and F046 have identical USB HAL-LL drivers and use the same USB library.

With regards,

ALABSTM, 2023-04-10:
Hi @DeckerCHAN,

I hope you are fine. Any feedback please? In the meanwhile, please allow me to close this issue as no activity has been registered since a while. We can easily reopen it should you have any updates to share. Thank you for your comprehension.

With regards,

closed 2023-04-10 by ALABSTM