STMicroelectronics/stm32f0xx-hal-driver #6: FLASH_OB_GetRDP has 32-bit return type but HAL_FLASHEx_OBErase implicitly converts it to 8-bits

jacobq, 2024-01-19:
**Describe the set-up**
 * Hardware agnostic (encountered while compiling program for STM32F072)
 * STM32CubeIDE v1.14.0 (build 19471_20231121_1200 (UTC))

**Describe the bug**
Compiling [`stm32f0xx_hal_flash_ex.c`](https://github.com/STMicroelectronics/stm32f0xx_hal_driver/blob/0c66a5bd9c5a1ba51f40d26f2387b3ee44b495d2/Src/stm32f0xx_hal_flash_ex.c#L105) results in (conversion related) warnings at build-time like this:
```
...
arm-none-eabi-gcc "../Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_rcc_ex.c" -mcpu=cortex-m0 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F072xB -c -I../Core/Inc -I../Drivers/STM32F0xx_HAL_Driver/Inc -I../Drivers/STM32F0xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F0xx/Include -I../Drivers/CMSIS/Include -Og -ffunction-sections -fdata-sections -Wall -Wconversion -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_rcc_ex.d" -MT"Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_rcc_ex.o" --specs=nano.specs -mfloat-abi=soft -mthumb -o "Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_rcc_ex.o"
../Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_flash_ex.c: In function 'HAL_FLASHEx_OBErase':
../Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_flash_ex.c:317:12: warning: conversion from 'uint32_t' {aka 'long unsigned int'} to 'uint8_t' {aka 'unsigned char'} may change value [-Wconversion]
  317 |   rdptmp = FLASH_OB_GetRDP();
      |            ^~~~~~~~~~~~~~~
../Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_flash_ex.c: In function 'HAL_FLASHEx_OBGetConfig':
../Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_flash_ex.c:449:23: warning: conversion from 'uint32_t' {aka 'long unsigned int'} to 'uint8_t' {aka 'unsigned char'} may change value [-Wconversion]
  449 |   pOBInit->RDPLevel = FLASH_OB_GetRDP();
      |                       ^~~~~~~~~~~~~~~
...
13:52:02 Build Finished. 0 errors, 2 warnings. (took 4s.966ms
```

**How To Reproduce**
1. Application agnostic.
2. Suspect is HAL ([`stm32f0xx_hal_flash_ex.c`](https://github.com/STMicroelectronics/stm32f0xx_hal_driver/blob/5fd31c52559a593dd9038c4abb4b85faeac728c6/Src/stm32f0xx_hal_flash_ex.c#L105))
3. No need to make any calls to reproduce -- problem is warning at compile time.
4. Can reproduce by compiling `stm32f0xx_hal_flash_ex.c` with conversion warnings enabled.

**Additional context**
Simply change the return type from `uint32_t` to `uint8_t` since result must be one of [`OB_RDP_LEVEL_0`, `OB_RDP_LEVEL_1`, or `OB_RDP_LEVEL_2`](https://github.com/STMicroelectronics/stm32f0xx_hal_driver/blob/5fd31c52559a593dd9038c4abb4b85faeac728c6/Inc/stm32f0xx_hal_flash_ex.h#L298-L301)

```patch
diff --git a/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_flash_ex.c b/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_flash_ex.c
index 7ea91b9..34068c7 100644
--- a/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_flash_ex.c
+++ b/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_flash_ex.c
@@ -102,7 +102,7 @@ static HAL_StatusTypeDef FLASH_OB_RDP_LevelConfig(uint8_t ReadProtectLevel);
 static HAL_StatusTypeDef FLASH_OB_UserConfig(uint8_t UserConfig);
 static HAL_StatusTypeDef FLASH_OB_ProgramData(uint32_t Address, uint8_t Data);
 static uint32_t          FLASH_OB_GetWRP(void);
-static uint32_t          FLASH_OB_GetRDP(void);
+static uint8_t           FLASH_OB_GetRDP(void);
 static uint8_t           FLASH_OB_GetUser(void);

 /**
@@ -899,7 +899,7 @@ static uint32_t FLASH_OB_GetWRP(void)
   *            @arg @ref OB_RDP_LEVEL_1 Read protection of the memory
   *            @arg @ref OB_RDP_LEVEL_2 Full chip protection
   */
-static uint32_t FLASH_OB_GetRDP(void)
+static uint8_t FLASH_OB_GetRDP(void)
 {
   uint32_t tmp_reg;
```

TOUNSTM, 2024-01-23:
See also [stm32f1xx_hal_driver#7](https://github.com/STMicroelectronics/stm32f1xx_hal_driver/pull/7)

jacobq, 2024-01-24:
If changing the signature is too much to ask, perhaps an explicit cast to suppress the warning would be OK?
https://github.com/STMicroelectronics/stm32f0xx_hal_driver/blob/5fd31c52559a593dd9038c4abb4b85faeac728c6/Src/stm32f0xx_hal_flash_ex.c#L317

```c
rdptmp = (uint8_t)FLASH_OB_GetRDP();
```

TOUNSTM, 2024-01-24:
Hello @jacobq,

Thank you for your report. We confirm the appearance of warnings by activating `-Wconversion warnings`. This problem has been reported to our development teams for further analysis. We'll get back to you as soon as we have any updates.

With Regards,