STMicroelectronics/stm32f0xx-hal-driver #7: Missing break-statement in LL_ADC_DeInit() can cause software to get stuck in a wait-loop

feaser, 2024-07-22:
**Describe the set-up**
 * Nucleo-F031K6
 * STM32CubeIDE 1.15.0 with GCC 12.3 Rel 1

**Describe the bug**
Function `LL_ADC_DeInit()` contains a wait loop, with timeout detection to make sure the software doesn't hang in case of an unexpected hardware state:

```c
  timeout_cpu_cycles = ADC_TIMEOUT_DISABLE_CPU_CYCLES;
  while (LL_ADC_IsDisableOngoing(ADCx) == 1U)
  {
    if(timeout_cpu_cycles-- == 0U)
    {
      /* Time-out error */
      status = ERROR;
    }
  }
```

The bug is that there is a missing `break;` statement, in case the timeout occurs. This causes the software to potentially still hang in this loop.

**How To Reproduce**
You can reproduce the problem:

1. Configure and enable the ADC1 on a channel to perform continuous conversions.
2. Call `LL_ADC_DeInit(ADC1);`

This causes the software to hang, because it is stuck in the before mentioned wait loop.

**Additional context**
Proposed solution: Add the `break;` statement:

```c
  timeout_cpu_cycles = ADC_TIMEOUT_DISABLE_CPU_CYCLES;
  while (LL_ADC_IsDisableOngoing(ADCx) == 1U)
  {
    if(timeout_cpu_cycles-- == 0U)
    {
      /* Time-out error */
      status = ERROR;
      break; /* <<< ADDED */
    }
  }
```

RJMSTM, 2024-07-24:
ST Internal Reference: 187367

KRASTM, 2024-08-02:
Fixed in commit https://github.com/STMicroelectronics/stm32f0xx_hal_driver/commit/1c14d36d88722f9cd7ed48d842b5ad07330008c7

closed 2024-08-02 by KRASTM