STMicroelectronics/stm32f0xx-hal-driver #9: USB buffer overflow on clear stall operation

pierre-x, 2025-01-20:
Inside `stm32f0xx_hal_pcd.c`, on normal operation, when the receive buffer is not full, the receive callback `CDC_Receive_FS` calls `USBD_CDC_ReceivePacket`, which re-arms the endpoint to receive new data and sets `ep->xfer_count` to zero.

In the case of a full buffer, the host may reach a timeout and request a Clear Stall. `USBD_LL_ClearStallEP` will call `HAL_PCD_EP_ClrStall` to re-arm the endpoint WITHOUT resetting `ep->xfer_count`, which will cause a buffer overflow when receiving the next packet.

IMHO, `HAL_PCD_EP_ClrStall()` should include an `ep->xfer_count = 0`

ALABSTM, 2025-01-30:
@pierre-x,

May I ask whether the use case you reported has been tested by any means and the described issue reproduced, or is it analysis based on code review?

Thank you,

ALABSTM, 2025-01-30:
ST Internal Reference: 201910

pierre-x, 2025-01-31:
> May I ask whether the use case you reported has been tested by any means and the described issue reproduced, or is it analysis based on code review?

It's a bug we have encountered, and that we can reproduce 100% of the time. We have corrected it as described in the bug report.

I can share by email some part of our source code if necessary.

ALABSTM, 2025-02-07:
Hi @pierre-x,

May I ask you whether you can share a sample project allowing to reproduce the issue?

Thank you,

ALABSTM, 2025-02-13:
Hi @pierre-x,

Any sample code or project to share please?

Thank you,

ALABSTM, 2025-02-28:
Hi @pierre-x,

Please allow me to close this thread as no reply since a while. Do not hesitate to log another one in case you believe the point is still relevant. Thank you for your comprehension.

With regards,